MitjaNemec/SaveRestoreLayout #2: Plugin not entering case: "SAVE"

Rene1695, 2022-05-03:
Hey,

i have been working for quite a long time with your plugin in KiCad 5. Now i tried to work with it in KiCad 6, but unluckly, i cant get it to work. Here is my setup and my Test-Project:

KiCad-Version:
[image]

KiCad-Test-Project:
Main-Sheet:
[image]

Hierachical Sheet:
[image]

> RefFrame is just a rectangle on the silkscreen layer.

PCB-Layout:
[image]

Now Steps i do to save the Layout inside the RefFrame:

1. Click and select the "RefFrame" U1 in PCBnew
2. Start your Plugin
3. Select "Save Layout"

After this, nothing happens anymore and no pckl file will be created. In my root directory of the project i find the log file which contains the following infos:

- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 149:Plugin executed on: 'win32'
- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 150:Plugin executed with python version: '3.9.10 (main, Feb 11 2022, 04:18:11) [MSC v.1928 64 bit (AMD64)]'
- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 151:KiCad build version: (6.0.2)
- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 152:Plugin version: 1.1.1
- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 153:Frame repr: <wx._core.Frame object at 0x000001808041FEE0>
- 05-03 12:10:00 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 172:Anchor footprint reference is 'U1'

I was travelling through your code and, to me, it seems like, that the plugin is not entering the case `if res == InitialDialog.SAVE:` because nothing else will be added into the log file. I tried to setup your plugin in PyCharm, bypassing the gui and set the res variable manually. Unfortunatly, this ends up in other errors, which might be caused by my interventions:

>  sheet_file = fp.GetProperty('Sheetfile')
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pcbnew.py", line 17716, in GetProperty
    raise KeyError("Property not found: " + key)
KeyError: 'Property not found: Sheetfile'

Do you have any kind of work-aorund for this? Im not sure, if the problem is on my side, or if i overlooked something.

Many thanks for your help and your work on this plugin!! 

Kind regards,
René

MitjaNemec, 2022-05-03:
Hi René,

Thanks for the feedback. Much appreciated.

1. Is the above complete contents of the .log file?
2. Can you attach the test project?
3. How did you install the plugin? Through PCM or through file? As is the plugin should not be allowed to install with 6.0.2 as it is compatible with 6.0.3 or higher (there is a bugfix in 6.0.3 which concerns the plugin)
4. Was the test project created in V5 or in V6?

Rene1695, 2022-05-03:
Hey,

thanks for the quick response! 

1. Yes, there is nothing else inside the log file (i attached it also inside the .zip)
2. Find the project attached
3. The plugin was installed through the PCM. I just upgraded KiCad from 6.0.2 to 6.0.4 (sorry i didnt checked that) and reinstalled the plugin through PCM but i have still the same result.
4. The project was created in Kicad 6.0.2. Do you have any concerns, that there will be issues when i comes to projects that are migrated to kicad 6?  

[Test_Snippet.zip](https://github.com/MitjaNemec/SaveRestoreLayout/files/8610165/Test_Snippet.zip)

MitjaNemec, 2022-05-03:
So, I've tested the project on my setup and it works. But I am working on linux, and you ran on Windows. As you already alluded it might be a case where for some reason wxWidgets don't behave the same across platforms (this happens quite often). I'll need to test and debug this on Windows and then I'll need to find a fix that works across platforms. Please hang in there

Rene1695, 2022-05-04:
Okey, thanks a lot! If i can help you in any kind of way, please feel free to let me know! I will try later on to set up a linux machine and test the plugin there, to confirm, that it is a windows issue.

MitjaNemec, 2022-05-04:
I've found the issue. On windows if the wxButton is set to anything other as wx.ID_ANY (e.g. wx.ID_OK or wx.ID_CANCEL) this id overrides any value that is returned via EndModal call.

For the time being you can change button ids in `initial_dialog_GUI.py` line 28 and 31 to wx.ID_ANY. It should work then. Official update should be available in a couple of days

Rene1695, 2022-05-04:
Very nice! Its working flawlessly, thank you very much! :)

closed 2022-05-08 by MitjaNemec