MitjaNemec/SaveRestoreLayout #4: Components without sheetfile stop plugin

NickKnatterton, 2022-08-16:
When restoring a plugin to a board with PCB-only components (e.g. mounting holes ) the plugin stops working. 

> 08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 149:Plugin executed on: 'win32'
08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 150:Plugin executed with python version: '3.9.10 (main, Jun 24 2022, 05:34:59) [MSC v.1932 64 bit (AMD64)]'
08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 151:KiCad build version: (6.0.6)
08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 152:Plugin version: 1.1.2
08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 153:Frame repr: <wx._core.Frame object at 0x00000161572E1160>
08-16 13:55:36 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 171:Anchor footprint reference is 'RefFrame3001'
08-16 13:55:37 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 235:Restore layout chosen
08-16 13:55:41 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 602:Getting board info
08-16 13:55:41 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 604:Get project schematics and layout data
08-16 13:55:41 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 132:getting a list of all footprints on board
08-16 13:55:41 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 251:Fatal error when creating an instance of RestoreLayout
Traceback (most recent call last):
  File "C:\X\plugins\com_github_MitjaNemec_SaveRestoreLayout\action_save_restore_layout.py", line 249, in Run
    restore_layout = RestoreLayout(board, dst_anchor_fp_ref)
  File "C:\X\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 605, in __init__
    self.prj_data = PrjData(self.board)
  File "C:\X\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 140, in __init__
    sheet_file = fp.GetProperty('Sheetfile')
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pcbnew.py", line 17747, in GetProperty
    raise KeyError("Property not found: " + key)
KeyError: 'Property not found: Sheetfile'

This seems to be a similar problem as mentioned in the replicate layout plugin [https://github.com/MitjaNemec/ReplicateLayout/issues/10] .
Catching the exception as you did in the replicate layout plugin works here as well, but I do not fully understand the logic of the following elif/else ( [(https://github.com/MitjaNemec/SaveRestoreLayout/blob/main/save_restore_layout.py#L146] ) construct to judge any side effects.

MitjaNemec, 2022-08-16:
The elif case is there only to cover the (now) unlikely case where project and partial layout was created in V5, then the layout was opened in V6 and plugin was invoked, bofore any refresh from schematics was executed. V5 did not hev sheetfile amd sheetname properties

MitjaNemec, 2022-08-16:
Thanks for reporting though. I'll look into it, but it'll be a while

commit 10571abd (2022-10-16): test for existense of Sheetfile and Sheetname properties in footprints.

Fixes: #4
commit 5085ac12 (2022-10-16): test for existense of Sheetfile and Sheetname properties in footprints.

Fixes: #4
closed 2022-10-16 by MitjaNemec