MitjaNemec/SaveRestoreLayout #8: [feature] hierarchical sheet name as default save name

T0jan, 2022-11-04:
hi,

Right now the save dialog uses the name of the project as default name of the `.pckl` file and I was thinking if it would make more sense to instead use the name of the hierarchical sheet the layout is based on. As soon as you want to save more then one layout part of a project you can't use the default name anymore anyway. Also the filename would then probably more descriptive of the function in case it isn't intended to be used right away or by another person.

So what I intend to have is for `my_awesome_project` with the hierarchical sheet `controller` the save dialog would open with `controller.pckl` as preselected filename and not with `my_awesome_project.pckl`.

In the end this would be just a small optimization (at least for me) so if it is too complicated to implement or interferes with other approaches this should be ignored.

Best Regards,
Tojan

MitjaNemec, 2022-11-04:
Thanks for the great suggestion. The change is in 1.1.5 which you can install from file (available under GitHub releases). It will take a while to become available in PCM.

closed 2022-11-04 by MitjaNemec

T0jan, 2022-11-07:
Thank you for implementing this that fast it will help me a lot :smile: