MitjaNemec/SaveRestoreLayout #9: plugin is crashing

jbforth, 2022-11-18:
Apologies if I'm doing something wrong, but the plugin is crashing.  Here is the log file.

---

11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 159:Plugin executed on: 'win32'
11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 160:Plugin executed with python version: '3.9.14 (main, Sep 19 2022, 05:11:57) [MSC v.1932 64 bit (AMD64)]'
11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 161:KiCad build version: (6.0.9)
11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 162:Plugin version: 1.1.5
11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 163:Frame repr: <wx._core.Frame object at 0x0000016C2558B550>
11-18 08:40:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 181:Anchor footprint reference is 'L1-2'
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 190:Save layout chosen
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 287:Working on C:/Lattice-FPGAs-KiCAD/Juke_audio_8_zone.kicad_pcb
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 132:getting a list of all footprints on board
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 292:Saving the current board temporary in order to leave current layout intact
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 298:Saving board as tempfile: C:\Users\bradf\AppData\Local\Temp\temp_board_file_for_save.kicad_pcb
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 303:Loaded temp boardfile: C:\Users\bradf\AppData\Local\Temp\temp_board_file_for_save.kicad_pcb
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 304:Get project schematics and layout data
11-18 08:40:25 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 132:getting a list of all footprints on board
11-18 08:40:50 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 237:Saving the layout in 'C:\\Lattice-FPGAs-KiCAD\\Regulator.pckl' for level 0
11-18 08:40:50 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 315:Saving layout for level: ['Power Regulator']
11-18 08:40:50 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 316:Calculating hash of the layout schematics
11-18 08:40:50 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 320:Saving hash for files: ['File: Regulator.kicad_sch']
11-18 08:40:50 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 246:Fatal error running SaveLayout
Traceback (most recent call last):
  File "C:\Users\bradf\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\action_save_restore_layout.py", line 239, in Run
    save_layout.save_layout(save_layout.src_anchor_fp.sheet_id[0:selected_level + 1], data_file,
  File "C:\Users\bradf\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 325, in save_layout
    md5hash = get_sch_hash(file_path, md5hash)
  File "C:\Users\bradf\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 65, in get_sch_hash
    with open(sch_file, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Lattice-FPGAs-KiCAD\\File: Regulator.kicad_sch'

MitjaNemec, 2022-11-18:
Thanks for the report.

This was already reported in #6. Basically this is due to the bug that was/is present in KiCad 6.0.8. You should use 6.0.9 and you need to update pcb from schematics in 6.0.9 in order to properly set the underlying data structures.

closed 2022-11-18 by MitjaNemec

jbforth, 2022-11-18:
Thanks for the quick reply, and for creating such a useful plugin.  I was running 6.09. I will see if the board was somehow saved in an older format.  I'm also updating KiCAD to a nightly build to see if that helps.  Thanks again!

MitjaNemec, 2022-11-18:
The plugins don't work under nighly versions (6.99). As for the bug you are experiencing the layout file was updated sometime in the past using 6.0.8. It is not enough to use 6.0.9, you have to update the layout from schematics using 6.0.9.