MitjaNemec/SaveRestoreLayout #11: Filename parsing cutting off characters

SimonMerrett, 2023-02-02:
This seems slightly similar to #6 but I can specifically see an error in the schematic file name that the plugin is failing to find:
```
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 172:Plugin executed on: 'win32'
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 173:Plugin executed with python version: '3.9.14 (main, Dec 23 2022, 23:36:19) [MSC v.1934 64 bit (AMD64)]'
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 174:KiCad build version: (6.0.11)
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 175:Plugin version: 1.1.8
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 176:Frame repr: <wx._core.Frame object at 0x000001B34A459430>
02-02 20:07:43 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 194:Anchor footprint reference is 'U901'
02-02 20:07:47 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 274:Restore layout chosen
02-02 20:07:55 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 75:Initialization start
02-02 20:07:55 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 77:Initialization done Group name:
02-02 20:08:20 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 648:Getting board info
02-02 20:08:20 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 651:Get project schematics and layout data
02-02 20:08:20 com_github_MitjaNemec_SaveRestoreLayout.save_restore_layout 132:getting a list of all footprints on board
02-02 20:08:20 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 299:Fatal error when creating an instance of RestoreLayout
Traceback (most recent call last):
  File "C:\Users\A\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\action_save_restore_layout.py", line 297, in Run
    restore_layout = RestoreLayout(board, dst_anchor_fp_ref, group_name)
  File "C:\Users\A\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 652, in __init__
    self.prj_data = PrjData(self.board)
  File "C:\Users\A\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 175, in __init__
    self.parse_schematic_files(self.sch_filename, schematic_found)
  File "C:\Users\A\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 209, in parse_schematic_files
    self.parse_schematic_files(sheetfile, dict_of_sheets)
  File "C:\Users\A\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 190, in parse_schematic_files
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: '5340A.kicad_sch'
```
`5340A.kicad_sch` is actually `Si5340A.kicad_sch`. 

KiCAD version info:
```
Application: KiCad Schematic Editor (64-bit)

Version: (6.0.11), release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 26 2023 07:02:30
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 38
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```
Any idea why it is cutting off schematic name first two characters?

MitjaNemec, 2023-02-03:
Thanks for reporting this.

I suspect that the issue is somewhere in line 205. `sheetfile = contents[j].lstrip("(property \"Sheet file\"").split()[0].replace("\"", "")`. My python string parsing skills are somewhat limited.  

Can you open the schematic file that has a subsheet pointing to `Si5340A.kicad_sch` and look for the `(sheet` section which refers to `Si5340A.kicad_sch` and past it here.

SimonMerrett, 2023-02-03:
```
  (sheet (at 86.36 41.91) (size 20.32 19.05) (fields_autoplaced)
    (stroke (width 0.1524) (type solid) (color 0 0 0 0))
    (fill (color 0 0 0 0.0000))
    (uuid 7cd25091-59df-478d-91a5-d499279c96e5)
    (property "Sheet name" "Clocks" (id 0) (at 86.36 41.1984 0)
      (effects (font (size 1.27 1.27)) (justify left bottom))
    )
    (property "Sheet file" "Si5340A.kicad_sch" (id 1) (at 86.36 61.5446 0)
      (effects (font (size 1.27 1.27)) (justify left top))
    )
  )
```
I should mention that I have since managed to get the restore layout to work. The problem I had was using a linked subordinate/hierarchical schematic sheet, rather than one with the file path inside the project folder for *one* of the projects only. I was effectively trying to share the schematic from the first design as well as the layout. I think the parsing error would have misnamed both instances and wouldn't have thrown an error if I had made my projects match properly. So I think it would be quite rare to find the plugin complaining about the name of the schematic sheet if the projects match properly.