MitjaNemec/SaveRestoreLayout #13: Saving failes when subsheets are in subdirectory

benrogmans, 2023-03-30:
It fails to save because the sheet can't be found. The reason is the the subsheet is actually located in my-project\sheets\subsheet.kicad_sch:

```
03-30 13:07:52 com_github_MitjaNemec_SaveRestoreLayout.action_save_restore_layout 259:Fatal error running SaveLayout
Traceback (most recent call last):
  File "C:\Users\ben\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\action_save_restore_layout.py", line 252, in Run
    save_layout.save_layout(save_layout.src_anchor_fp.sheet_id[0:selected_level + 1], data_file,
  File "C:\Users\ben\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 385, in save_layout
    md5hash = get_sch_hash(file_path, md5hash)
  File "C:\Users\ben\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_SaveRestoreLayout\save_restore_layout.py", line 65, in get_sch_hash
    with open(sch_file, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\ben\\Documents\\my-project\\subsheet.kicad_sch'
```

MitjaNemec, 2023-03-31:
Thanks for reporting this.

Can you share your project? If you can't, can you maybe prepare and share a test project with which you can recreate the issue?

MitjaNemec, 2023-04-03:
@benrogmans 

I've just published a fix for one potential issue when handling schematics filename paths. Hope this covers your case. The updated version should be available in PCB within couple of days, but you can install as a file if you take it from [releases](https://github.com/MitjaNemec/SaveRestoreLayout/releases)

MitjaNemec, 2023-10-11:
It looks like this is the same as #15 which should be fixed