MitjaNemec/SaveRestoreLayout #17: Restoring a loyout failed with "Source and destination schematics don't match!"

worldwidemv, 2023-10-06:
Hello,

thanks for the plugin, it looks very promising.
Unfortunately, it does not work for me under Linux with KiCad 7.0.7

It looks like you compare hashes of the two schematic files, and this fails because the `instances` section of every symbol instance in the schematic is updated as soon as I copy the hierarchical sheet into the new project. 

So, if I make a diff for the two sheets:
* the original in the project `AnalogPower_In5V_Out5V-+2V5`; sheet name `M_AnalogePower.kicad_sch`
* the copy in the project `RedPitaya_PD_HAT_V1`; sheet name `M_AnalogePower.kicad_sch`
I see:
```
  (symbol (lib_id "Device:C") (at 83.82 83.82 0) (unit 1)
    (in_bom yes) (on_board yes) (dnp no)
    (uuid f4de520c-a4ed-4fc0-bcf6-e54e8493a81f)
    (property "Reference" "C3" (at 83.82 81.28 0)
      (effects (font (size 1.27 1.27)) (justify left))
    )
    (property "Value" "1u/25V" (at 87.63 86.36 0)
      (effects (font (size 1.27 1.27)) (justify right))
    )
    (property "Footprint" "Capacitor_SMD:C_0603_1608Metric" (at 84.7852 87.63 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "Datasheet" "~" (at 83.82 83.82 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (pin "1" (uuid 1844842e-e0ed-498a-b18e-a6c52239d0ac))
    (pin "2" (uuid a29a2047-2320-4859-bad6-8b2ab71d4a77))
    (instances
+      (project "RedPitaya_PD_HAT_V1"
+        (path "/3d3375cd-4de6-4078-82fa-39d0f373d068/fc1a81c4-d7d5-4cfe-92fb-80830ab2460d"
+          (reference "C3") (unit 1)
+        )
+      )
      (project "AnalogPower_In5V_Out5V-+2V5"
        (path "/53854ffa-83d7-44ca-aa23-c38aa4123b5e/25b7b422-ee30-4321-8b2d-215bafc28d1d"
          (reference "C3") (unit 1)
        )
      )
    )
  )
```
I am sot sure if that is new in KiCad 7, but it breaks the plugin unfortunately.

## Workflow

1. Copy the schematic of the hierarchical sheet from the module project to the new project.
2. Open the new project and the module project and copy the hierarchical sheet block from the module project and past the block into the new projects schematic.
3. Go to new projects PCB and update the board.
4. Select a part and run the "Save/Restore Layout" plugin; chose restore ....

No editing the copied schematic, re-annotation parts or anything like that.

Here is the [save_restore_layout.log](https://github.com/MitjaNemec/SaveRestoreLayout/files/12832567/save_restore_layout.log)

## KiCad Info

Application: KiCad x86_64 on x86_64

Version: 7.0.7-7.0.7~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16

Platform: KDE neon 5.27, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: Aug 13 2023 23:14:49
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

MitjaNemec, 2023-10-09:
Hi, thanks for the thorough bug report. It looks like schematics saving was changed a bit during the 7.0.x releases. IIRC I've already seen something on this by Wayne (lead KiCad dev) but I haven't put two and two together that this would impact my plugins. Anyhow, I already ignore some things when calculating the hashes, so I'll have to additionally exclude the "instances" sections.

Again thanks for the thorough bug report. It dose help a lot and it will help with the bugfix

worldwidemv, 2023-10-09:
My pleasure. I think giving you as much information as possible is the least I can do....
Especially considering the time/work/convenience you are going to save me :-) Thanks again...

For other affected users of 7.0: Commenting line  744 and 745 in `save_restore_layout.py` works as a workaround.

MitjaNemec, 2023-10-11:
This should be fixed in [2.1.1](https://github.com/MitjaNemec/SaveRestoreLayout/releases/tag/2.1.1). Can you please install it from file and test it before I push it to be available via PCM.

worldwidemv, 2023-10-12:
Works like a charm. Thanks.

closed 2023-10-12 by worldwidemv